ArduPilot/SiK #25: The SiK Firmware documentation should be centralized

LorenzMeier, 2015-10-17:
Right now its replicated way too often:
http://copter.ardupilot.com/wiki/common-3dr-radio-version-2/
http://copter.ardupilot.com/wiki/common-3dr-radio-v1/
http://copter.ardupilot.com/wiki/common-3dr-radio-advanced-configuration-and-technical-information/
http://copter.ardupilot.com/wiki/common-rfd900/
https://pixhawk.org/peripherals/radio-modems/start?s[]=3dr&s[]=radio#mhz868_mhz900_mhz_long_rangeoutdoor
http://plane.ardupilot.com/wiki/common-rfd900/
http://plane.ardupilot.com/wiki/common-3dr-radio-v1/

MauroMombelli, 2016-06-29:
some of this link and some on the readme (http://ardupilot.org/copter/docs/common-optional-hardware/common-telemetry-landingpage/common-3dr-radio-advanced-configuration-and-technical-information.html) are not available anymore

kaklik, 2020-01-21:
Hi, 
none of links above works for now..