ArduPilot/SiK #27: LBT fix is not released so cannot conform to new EU regulations

CodeChief, 2016-02-24:
SiK already supports LBT I see and an issue failing to save the setting appears to be fixed in one of the last commits c264c6766ddf778e75fe0b901635ace90b0836f7 on October 17th 2015. But the beta and stable firmware released at http://firmware.diydrones.com are still from 2014.

Can somebody who already compiled this firmware please copy the binaries up to the DIY drones site, and if tested successfully consider making it "stable" so we are not breaking any new regulations (note even the FrSky firmware has been completely updated with new warnings on old models to update to LBT capable firmware).

tridge, 2016-06-05:
i'm doing a new release now, beta is already out
please let me know if it doesn't fix it for you

closed 2016-06-05 by tridge

CodeChief, 2016-06-05:
Just tested on the bench and the new firmware loads successfully (did not crash after setting LBT > 0) and will successfully establish a link when LBT > 100. Not sure what the correct outdoor settings are for LBT but it appears to be working and is something we can now tune. For information my hardware is the HK 433MHz radio kit (clone) both full USB and micro APM versions. Thanks!