ArduPilot/SiK #40: Crosstalk between different netids

mantelt, 2017-06-08:
Using the current master, I noticed that I get some of the (mavlink) packets sent from a device with different netid. 

Having 3 radios A (`netid = 25`, plugged to pixhawk), B (`netid = 25`) and C (`netid = 15`) (see complete config below), communication between A and B seems to work fine. Plugging radio C, I receive quiet a few of the mavlink packets sent by A.

Using the old release-1-9, I didn't notice such a behavior. 

#### Radio A ####
```
ATI
RFD SiK 2.0 on HM-TRP
ATI5
S0:FORMAT=26
S1:SERIAL_SPEED=57
S2:AIR_SPEED=64
S3:NETID=25
S4:TXPOWER=20
S5:ECC=1
S6:MAVLINK=1
S7:OPPRESEND=0
S8:MIN_FREQ=433050
S9:MAX_FREQ=434790
S10:NUM_CHANNELS=10
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=0
S15:MAX_WINDOW=131
```

#### Radio B ####
```
ATI
RFD SiK 2.0 on HM-TRP
ATI5
S0:FORMAT=26
S1:SERIAL_SPEED=57
S2:AIR_SPEED=64
S3:NETID=25
S4:TXPOWER=20
S5:ECC=1
S6:MAVLINK=1
S7:OPPRESEND=1
S8:MIN_FREQ=433050
S9:MAX_FREQ=434790
S10:NUM_CHANNELS=10
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=0
S15:MAX_WINDOW=131
```

#### Radio C ####
```
ATI
RFD SiK 2.0 on HM-TRP
ATI5
S0:FORMAT=26
S1:SERIAL_SPEED=57
S2:AIR_SPEED=64
S3:NETID=15
S4:TXPOWER=20
S5:ECC=1
S6:MAVLINK=1
S7:OPPRESEND=0
S8:MIN_FREQ=433050
S9:MAX_FREQ=434790
S10:NUM_CHANNELS=10
S11:DUTY_CYCLE=100
S12:LBT_RSSI=0
S13:MANCHESTER=0
S14:RTSCTS=0
S15:MAX_WINDOW=131
```

Serveurperso, 2017-06-08:
Do you truly receive packets (or corrupted data) from other network ID on terminal or you just check the RX led on the radio ?
With ECC enabled there is a software GOLAY function just check for ID and reject packet.

With a setup where you share same hopping frequency and speed you can get some packet loss

```
// setup a 16 bit network ID
//
void
radio_set_network_id(uint16_t id)
{
        netid[0] = id&0xFF;
        netid[1] = id>>8;
        if (!feature_golay) {
                // when not using golay encoding we use the hardware
                // headers for network ID
                register_write(EZRADIOPRO_TRANSMIT_HEADER_3, id >> 8);
                register_write(EZRADIOPRO_TRANSMIT_HEADER_2, id & 0xFF);
                register_write(EZRADIOPRO_CHECK_HEADER_3, id >> 8);
                register_write(EZRADIOPRO_CHECK_HEADER_2, id & 0xFF);
        }
}
```

```
        // decode the header
        errcount = golay_decode(6, buf, gout);
        if (gout[0] != netid[0] ||
            gout[1] != netid[1]) {
                // its not for our network ID 
                debug("netid %x %x\n",
                       (unsigned)gout[0],
                       (unsigned)gout[1]);
                goto failed;
        }
```

I can make tests with my fork because I can use all raw radio features from the last version of SiK without hopping/tdd and mavlink. (transparent serial single frequency modem)

Serveurperso, 2017-06-09:
You are right: there is no bad net id packet rejection with ECC (soft golay) enabled.

Without ECC the LED light up (radio_receive_in_progress) but is rejected.

I look if I can find the bug...

Serveurperso, 2017-06-09:
I found the problem. The golay ECC code is never included because developpers added AES support on some board type, and disabled ECC on these boards.

golay23.h :
```
#ifndef INCLUDE_AES
#define INCLUDE_GOLAY
#endif
```
This is the only one "#define INCLUDE_GOLAY" from the code but it look NOT suffisient.

I just added this "#define INCLUDE_GOLAY" inside board.h to workaround for my HM_TRP board and it's work now: there is no more crosstalk:D

```
(root|~/SiK.Serveurperso/Firmware/radio) git diff
diff --git a/Firmware/include/board.h b/Firmware/include/board.h
index 7e29b32..8219bf1 100644
--- a/Firmware/include/board.h
+++ b/Firmware/include/board.h
@@ -97,6 +97,7 @@
 #if   defined(BOARD_rf50)
 # include "board_rf50.h"
 #elif defined(BOARD_hm_trp)
+#define INCLUDE_GOLAY
 # include "board_hm_trp.h"
 #elif defined(BOARD_rfd900)
 # include "board_rfd900.h"
```
This issue is now "Regression : Golay ECC code is never included on other boards"

Serveurperso, 2017-06-09:
The clean solution is just to add this inside radio.h like on golay23.h :
```
#ifndef INCLUDE_AES
#define INCLUDE_GOLAY
#endif
```
https://github.com/Serveurperso/SiK/commit/0f5f65c8848685bcf7ef5bb8915da3fc4c584d39

mantelt, 2017-06-12:
Hey @Serveurperso 
Thanks a lot for debugging and fixing this so quickly! I will re-check your solution hopefully soon (I'm quite busy at the moment), but as you already checked it I'm positive that this will just work now.

Serveurperso, 2017-06-12:
You must apply this patch https://github.com/Serveurperso/SiK/commit/0f5f65c8848685bcf7ef5bb8915da3fc4c584d39 or developers must add this commit (only) inside the official repo.