ArduPilot/SiK #44: SiK multipoint on 3DR telemetry modules

mzahana, 2018-06-19:
Hi.

Is it possible to flash 3DR telemetry modules 433MHz with SiK version that supports multipoint?
If yes, what is the procedure ?

Thanks.

Rusty105, 2018-06-25:
By multipoint what do you mean? multiple radios with the same Net ID, or multiple radios across multiple Net IDs?

mzahana, 2018-06-25:
I guess it works. I just tried it. There is a branch for SiK which supports multipoint. I flashed it and it worked. The multipoint one allows me to communicate with multiple modules in the air using only one ground module.

sadathhussain, 2018-08-16:
Hi,

Can you please post links to the firmware that supports multi point. I found one link that says I need a SiLabs Debug adapter to flash the firmware. Is this necessary?? Also, did you flash the firmware using Mission Planner?

AlexisTM, 2018-10-16:
@sadathhussain You don't need the SiLabs Debug adapter, you can flash it via USB; With the make alias

kurbi, 2019-05-25:
Hi
I'm trying to built multipoint network with hm-trp module and 2.5 firmware. Topology is: base (id:0, bc:65535), relay (id:1, bc:65535), drones (id:2-4 bc:1).
I have tried a lot of configurations, but none of them seems to provide stable telemetry and transferring waypoints is a no go. Flow controll is disabled due to glitch that makes radios go to configure mode (red light) when the wires are connected. 
Does anyone managed to make multipoint network with relay work? Your configuration?

prem48science, 2019-08-02:
> I guess it works. I just tried it. There is a branch for SiK which supports multipoint. I flashed it and it worked. The multipoint one allows me to communicate with multiple modules in the air using only one ground module.

can you please share, how you connected multiple vehicles using one ground module in mission planner.

Jai-GAY, 2023-04-17:
> multiple vehicles using one ground module in mission planner

https://ardupilot.org/planner/docs/common-connect-mission-planner-autopilot.html#connecting-to-multiple-vehicles

masmooh, 2023-08-24:
hi https://github.com/mzahana,

Could you please share step by step to flash silabs 1000 for multipoints including requirements files?

Thank you,