ArduPilot/SiK #45: Which toolchain should I use

e135193, 2018-07-02:
Hello,

     I want to compile the source code. Which toolchain should I use ? could someone provide a link ?

Serveurperso, 2018-07-02:
Hi, if I remember it need g++ and sdcc (+ git and make lol)

e135193, 2018-07-02:
thanks. it works well

spth, 2019-10-21:
It probably works with some rather old SDCC out-of-the-box.

Using current SDCC, the firmware compiles for me using this patch:

~~~~
diff --git a/Firmware/bootloader/product.mk b/Firmware/bootloader/product.mk
index d3b9f5d..9d647ea 100644
--- a/Firmware/bootloader/product.mk
+++ b/Firmware/bootloader/product.mk
@@ -34,7 +34,7 @@ PRODUCT_DIR	:=	$(patsubst %/,%,$(dir $(lastword $(MAKEFILE_LIST))))
 PRODUCT_INSTALL	 =	$(foreach frequency,$(FREQUENCIES), $(OBJROOT)/$(PRODUCT)~$(frequency).hex)
 
 CFLAGS		+=	-DBL_VERSION=$(VERSION)
-CFLAGS		+=	--model-small --no-xinit-opt --opt-code-size --Werror
+CFLAGS		+=	--model-small --no-xinit-opt --opt-code-size
 #CFLAGS		+=	--fverbose-asm
 
 # Set limits for the low (CSEG) and high (HIGHCSEG) code segments.
diff --git a/Firmware/radio/product.mk b/Firmware/radio/product.mk
index cf21a15..74880ad 100644
--- a/Firmware/radio/product.mk
+++ b/Firmware/radio/product.mk
@@ -34,7 +34,7 @@ PRODUCT		 =	radio~$(BOARD)
 PRODUCT_DIR	:=	$(patsubst %/,%,$(dir $(lastword $(MAKEFILE_LIST))))
 
 CFLAGS		+=	-DAPP_VERSION_HIGH=$(VERSION_MAJOR) -DAPP_VERSION_LOW=$(VERSION_MINOR)
-CFLAGS		+=	$(CFLAG_MODEL) --opt-code-speed --Werror --std-sdcc99 --fomit-frame-pointer
+CFLAGS		+=	$(CFLAG_MODEL) --opt-code-speed --std-sdcc99 --fomit-frame-pointer
 #CFLAGS		+=	--fverbose-asm 
 PRODUCT_SUPPORT_BANKING = 1
 INC_DIR_AES	 = AES
diff --git a/Firmware/radio/serial.c b/Firmware/radio/serial.c
index 1320273..ac80aeb 100644
--- a/Firmware/radio/serial.c
+++ b/Firmware/radio/serial.c
@@ -537,12 +537,13 @@ serial_read_space(void)
 	return space;
 }
 
-void
-putchar(char c) __reentrant
+int
+putchar(int c) __reentrant
 {
 	if (c == '\n')
 		_serial_write('\r');
 	_serial_write(c);
+	return c;
 }
 ~~~~