ArduPilot/SiK #50: Firmware Installation Error.

SamwelOpiyo, 2019-02-24:
On running a `make install` in the Firmware folder on an OSX, I get:

```
% build bootloader for hm_trp
obj/hm_trp/bootloader~hm_trp/bootloader.dep:1: *** missing separator.  Stop.
make: *** [build~bootloader~hm_trp] Error 2
```

Looks like it is some Makefile formatting issue but I could not lay may hands on it.

SamwelOpiyo, 2019-02-24:
On deleting obj directory created under firmware and running `make install again` I run into:

```
% build bootloader for hm_trp
CC bootloader/bootloader.c
bootloader/bootloader.c:102: error 158: overflow in implicit constant conversion
make[1]: *** [obj/hm_trp/bootloader~hm_trp/bootloader.rel] Error 1
make: *** [build~bootloader~hm_trp] Error 2
```

I have tried the installation on linux too and it raises:

```
% build bootloader for rfd900a
make[1]: Entering directory 'SiK/Firmware'
AS bootloader/crtstart.asm
CC bootloader/bootloader.c
bootloader/bootloader.c:102: error 158: overflow in implicit constant conversion
make[1]: *** [include/rules.mk:135: obj/rfd900a/bootloader~rfd900a/bootloader.rel] Error 1
make[1]: Leaving directory 'SiK/Firmware'
make: *** [Makefile:95: build~bootloader~rfd900a] Error 2
```

thedemz, 2019-02-27:
The Linux error seems to involve the SDCC compiler.

https://sourceforge.net/p/sdcc/discussion/1865/thread/759a466b/

I tried to change the code in bootloader.c

```
    i = 0;
    for (; i < 255; i++) {
        if (BUTTON_BOOTLOAD == BUTTON_ACTIVE)
            debounce_count++;                   
    }
```

Then got the compile error:

```
% build bootloader for rfd900a
make[1]: Entering directory '/SiK/Firmware'
CC bootloader/bootloader.c
bootloader/bootloader.c:183: error 126: unreachable code
make[1]: *** [include/rules.mk:123: obj/rfd900a/bootloader~rfd900a/bootloader.rel] Error 1
make[1]: Leaving directory '/SiK/Firmware'
make: *** [Makefile:95: build~bootloader~rfd900a] Error 2
```

```
sdcc --version

SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.8.0 #10562 (Linux)
published under GNU General Public License (GPL)
```

Lets try an older version of SDCC maybee

thedemz, 2019-02-27:
```
https://github.com/RFDesign/SiK

git checkout v1.11

sdcc --version

SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/pic16/pic14/TININative/ds400/hc08/s08/stm8 3.5.0 #9253 (Feb 27 2019) (Linux)
published under GNU General Public License (GPL)
```

Able to compile with version 3.5.0 of sdcc

https://sourceforge.net/projects/sdcc/files/sdcc/3.5.0/

thedemz, 2019-02-27:
Note that it is Python 2 that is required!

Files that need changes:

```
Firmware/tools/hexpatch.py
Firmware/tools/check_code.py
```

change the code

```
#!/usr/bin/env python
```

to

```
#!/usr/bin/env python2
```

Then command

```
make install
```

Should work

SamwelOpiyo, 2019-02-28:
Thank you. That worked.

closed 2019-02-28 by SamwelOpiyo