ArduPilot/SiK #55: Hacktoberfest: Python 3 compatibility issues

cclauss, 2019-10-03:
[flake8](http://flake8.pycqa.org) testing of https://github.com/ArduPilot/SiK on Python 3.7.1

$ __flake8 . --count --select=E9,F63,F7,F82 --show-source --statistics__
```
./Firmware/tools/check_code.py:17:13: E999 SyntaxError: invalid syntax
    print sys.argv
            ^
./Firmware/tools/bank-alloc.py:143:31: E999 SyntaxError: invalid syntax
					print "Failed to allocate", module[0], "with size", module[1], \
                              ^
./Firmware/tools/update_mode.py:35:12: F821 undefined name 'pexpect'
    except pexpect.TIMEOUT:
           ^
./Firmware/tools/atcommander.py:269:81: E999 SyntaxError: invalid syntax
        print "Please specify a --list-* or --set-* operation (try -h if unsure)"
                                                                                ^
./Firmware/tools/rssi.py:51:31: E999 TabError: inconsistent use of tabs and spaces in indentation
            	sys.stdout.flush()
                              ^
./Firmware/tools/uploader.py:100:31: E999 TabError: inconsistent use of tabs and spaces in indentation
                if use_mavlink:
                              ^
5     E999 SyntaxError: invalid syntax
1     F821 undefined name 'pexpect'
6
```
__E901,E999,F821,F822,F823__ are the "_showstopper_" [flake8](http://flake8.pycqa.org) issues that can halt the runtime with a SyntaxError, NameError, etc. These 5 are different from most other flake8 issues which are merely "style violations" -- useful for readability but they do not effect runtime safety.
* F821: undefined name `name`
* F822: undefined name `name` in `__all__`
* F823: local variable name referenced before assignment
* E901: SyntaxError or IndentationError
* E999: SyntaxError -- failed to compile a file into an Abstract Syntax Tree

monkeypants, 2019-10-09:
The quickest way to fix this would be to state in the README that python 3 is not yet supported.

Are these 6 things the only changes needed for python3 support?

cclauss, 2019-10-10:
83 days until Python 2 end of life.  These changes are necessary but probably not sufficient for a complete port to Python 3.  It is not best practice to port to Python 3 in a single pull request unless the codebase is trivially small.