ArduPilot/SiK #57: conflict with previous declaration of 'putchar'

pklapperich, 2019-10-10:
Unable to build with sdcc 3.9.0:

```
CC radio/golay.c
CC radio/serial.c
radio/serial.c:541: error 98: conflict with previous declaration of 'putchar' for attribute 'type' at /usr/bin/../share/sdcc/include/stdio.h:86
from type 'int function ( int fixed) fixed'
  to type 'void function ( unsigned-char fixed) __reentrant fixed'
make[1]: *** [include/rules.mk:135: obj/rfd900a/radio~rfd900a/serial.rel] Error 1
make[1]: Leaving directory '/home/paulk/botlink/SiK/Firmware'
make: *** [Makefile:95: install~radio~rfd900a] Error 2
```

I changed the signature in serial.c to 
```
int                                                                                                 
putchar(int c) __reentrant
```

and it's working fine for me. But I suspect there's a more correct fix.

I found [what looks like the same issue](https://sourceforge.net/p/sdcc/mailman/message/12702250/) on the sdcc mailing list from 2007, but the PUTCHAR definition doesn't exist, so the reply there was unhelpful.

spth, 2019-10-21:
This should be correct:

~~~~
int
putchar(int c) __reentrant
{
	if (c == '\n')
		_serial_write('\r');
	_serial_write(c);
	return c;
}
~~~~

By the C standard, putchar() takes an int, and returns it (using an int allows for error reporting, by returning EOF).

AndKe, 2019-10-21:
Thank you, this stopped me too.
With SDCC 3.8 , I needed to remove "-Werror"  to get it compiled, - did you need it removed with 3.9 ?

spth, 2019-10-21:
I had to remove it for SDCC 3.9.5.