ArduPilot/SiK #59: Using two pairs of Sik devices in the same area

ayavilevich, 2020-05-18:
Hi, would appreciate to get your opinion on the best way to configure the Siks in the presence of other pairs in the immediate area.

Approach 1:
Divide the frequency range into several distinct (and smaller) bands and allocate each pair with a unique range. Naturally also give unique net ids to each pair.

Approach 2:
Use the same range for all pairs, possibly with more channels, and rely on net id for conflict resolution.

btw, what would be the smallest channel width allowed?

Thanks.

kaklik, 2021-05-15:
Hello,  how many modem pairs in the same area are you interested in? 
We reliably operate the two modem pairs in the same area with different NETIDs only.

MrLAtt, 2023-02-17:
@kaklik Sorry for bringing this up in an old issue, but I have also been trying to use multiple SIk radios (RFD868) in the same area and have been having problems. 

Could you specify more about the setup that you've got working? 

Particulalrly any settings the you changed from defaults, the physical seperation of the ground units to each other, and the air units to each other, power output settings and antennas?

The most success I have had is with all default settings (apart from NetIDs), the radios on the ground seperated by about 4 meters (couldn't go any further) omnidirectional antennas all round (what I had available) and turning down the power on the radios to 100mw (down from 1w all round). this works somewhat ok, but range is significantly reduced, and it becomes dificult to download parameters

kaklik, 2023-02-17:
> @kaklik Sorry for bringing this up in an old issue, but I have also been trying to use multiple SIk radios (RFD868) in the same area and have been having problems.
>
> Could you specify more about the [...]

Hi, I am pretty sure that issue is what you experiencing is caused by the poor RF spectral parameters of RFD868.  It does not relate to the firmware or firmware setup. The RFD868 "1 Watt" capability is a catalog parameter number, which could not be effectively used in combination with other parameters like multiple devices at the same location separated by different NetIDs. 

The setup which we are using is our [TFSIK01 modems](https://github.com/ThunderFly-aerospace/TFSIK01) which we developed to overcome the poor hardware quality of modems available on market.

ayavilevich, 2023-02-17:
FWIW, I had two pairs and I ended up separating them on the frequency range as much as I could. Only then I could achieve decent performance. I have to note that I had both sending a lot of data, so it was not a sporadic packet of data here and there but something more similar to 10hz telemetry.
Don't remember which particular hardware I was using. It was something common.