ArduPilot/SiK #78: Failing to connect to SiK radio

vijaykoundinya, 2022-06-16:
failed to run uploarder.py

kaklik, 2022-11-06:
Do you have some debug output to allow diagnostics of that problem?

DaniKaganovich, 2023-05-31:
Also failed to connect.
On both modules the same firmware.
If ground module connected to PC and airmodule connected to Pixhawk 6C it begin lighting solid green after power on. at the same time ground module is blinking green. If I change Net ID on ground module to another (from 25 to 26) and save air module also become blinking green. In any case after copy to remote its not connecting

kaklik, 2023-05-31:
> Also failed to connect. On both modules the same firmware. If ground module connected to PC and airmodule connected to Pixhawk 6C it begin lighting solid green after power on. at the same time [...]

As I understand the original point of the issue was a malfunction of uploarder.py In case you have another problem with SiK firmware, please open a different issue containing the detailed description and procedure to replicate the problem. Thanks.