ArduPilot/SiK #91: Does anyone have EDA schematics for a radio or a surface-mountable sik radio module?

factoidforrest, 2025-01-08:
Hi there,
I'm trying to embed a sik radio in my AUV robot for long range telemetry in a compact size. I'd love to embed onto my custom-pcb. Ideally I'd love to buy or find an smd radio module I could mount to the PCB. I could also build one into the board including all the SMD components but I can't find any schematics for the hardware. Is the hardware open source?

Id love to buy a module like this, that would be the easiest: https://dwmzone.com/en/data-link-rf-module/135-1649-3dr-radio-telemetry-433mhz-868mhz-915mhz-module-kit.html#/23-center_frequency-915mhz/653-type-3dr_module

I can't quite tell if this store is selling the raw board and their shipping is quite insane.

factoidforrest, 2025-01-09:
Oh heres an open source hardware design:

https://github.com/ThunderFly-aerospace/TFSIK01

kaklik, 2025-03-29:
> Oh here is an open source hardware design:
> 
> https://github.com/ThunderFly-aerospace/TFSIK01

Yes, exactly!
 We are from ThunderFly, and we designed the TFSIK01 hardware from the ground up. Since we are the original developers, we can also provide support and additional services. More details here: [TFSIK01 Documentation](https://docs.thunderfly.cz/avionics/TFSIK01/).